shiffman/OpenKinect-for-Processing #10: "A library used by this sketch is not installed properly."

mrengy, 2015-01-13:
Thanks for the contribution. Maybe you can help me sort this out?

I got this error when trying to run the sample script described at http://shiffman.net/p5/kinect/

I'm running Windows 8, Java 1.8.0.25, Processing 2.2.1

I double checked that both OpenKinect and SimpleOpenNI libraries were installed in my Processing folder.

I came to http://forum.processing.org/two/discussion/2805/issues-with-openkinect/p1#Comment_14308 
, which seemed to be close to my problem, and I followed the link to https://github.com/shiffman/OpenKinect-for-Processing , which says in the Readme, "This library is (hopefully temporarily) broken. However, it can work if you install libusb separately."

What should I do here? Is the library still broken, or should I try to install it anyway?

I don't think I can use homebrew since I'm on Windows.

fwasim, 2015-01-25:
Have you installed it anyway? Did it work?

shiffman, 2015-01-25:
This library does not support windows unfortunately.

fwasim, 2015-01-25:
I am planning to install it on ubuntu 12.04. Is this still broken? I already have libusb installed.

shiffman, 2015-01-25:
It should work I think!  Haven't been able to test on linux recently.   Right now it only works with 1414 I believe, but still hoping / planning to do a revamp soon for 1473 and v2.  The official MS SDK is a better way to go for v2, but it's windows only so this will hopefully be a solution for at least getting raw depth.

fwasim, 2015-01-25:
Thanks. I'll give it a try and update.

mrengy, 2015-01-27:
Thanks for the response. Maybe others would benefit from clarifying this in the readme in Github and also on http://shiffman.net/p5/kinect/ . I did see in the latter link the line "Here's a quick list of what is out there and what is supported in Processing for Mac OS X.", but I didn't get the message that it was Mac OS (and maybe Linux) ONLY if that was the intent, especially since there was no mention of Mac OS or Linux under "What hardware do I need?".

shiffman, 2015-01-27:
Indeed you are right I need to make this more clear.  Re-titling this issue and re-opening.

shiffman, 2015-07-02:
Make sure this is covered during #15.

arky, 2016-07-24:
OS Update: Ubuntu 16 (64bit) with Kinect v1 (version 1414). Am happy to report that Processing 3 and Openkinect for Processing library works out the box.  

Please let me know if I could help doing testing on Ubuntu 32/64  Windows 7/8 machines. @shiffman @ThomasLengeling 

*OS LSB Version:    core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.1 LTS
Release:        16.04
Codename:       xenial

* Processing 3.1.1 (Linux) 

* Open Kinect for Processing 1.0 (Installed Contributed library via processing IDE)
Daniel Shiffman and Thomas Sanchez  A Mac OS X Kinect implementation using open source drivers (libfreenect).