shiffman/OpenKinect-for-Processing #11: Ability to restart Kinect tracking

Jickelsen, 2015-03-31:
I might attempt to implement this myself and submit a pull request, but I wanted to check first if there is any way to reinitialize the Kinect tracker through the existing API, without restarting the sketch.

I've tried calling Kinect.quit() and creating a new instance of the object, but it usually ends up giving null pointer exceptions. 

The reason I want to do this in the first place is that I'm creating an unsupervised exhibition running a tracking routine, but I'm having reoccurring tracking freezes and a quick fix would be to reinitialize the tracking. I realize it may be libusb-related but I have yet to find another fix.

shiffman, 2015-04-01:
The library does not include this feature at the moment.  To be honest this library is woefully out of date.  I am hopeful this will become a project this summer and lots of new features and fixes will be forthcoming.

shiffman, 2015-07-24:
This is an issue from the old library.  If this feature is still needed for the new one, please re-open.

closed 2015-07-24 by shiffman