shiffman/OpenKinect-for-Processing #12: openkinect2 not finding libturbojpeg.dylib

shiffman, 2015-07-02:
When I run the library on a fresh machine without libfreenect2 and related libraries installed separately, I'm getting an error (edited for clarity):

```
dlopen(/Users/.../Processing/libraries/openKinect2/library/libJNILibfreenect2.dylib,1):
Library not loaded: @executable_path/../Frameworks/libturbojpeg.dylib Referenced from:
/Users/.../Processing/libraries/openKinect2/library/libJNILibfreenect2.dylib
```

Looks like is getting `libJNILibfreenect2.dylib` ok, but that library references `libturbojpeg.dylib` not in a local path?

Looping in @ThomasLengeling

ThomasLengeling, 2015-07-02:
OK got, when i was doing the installation I did a small hack and copied libturbojpeg.dylib to /usr/lib folder, because I was having that error, I forgot about that, let me fix it.

ThomasLengeling, 2015-07-02:
Ok now I remember why  ```libturbojpeg.dylib``` is int the /usr/ lib folder

The libturbojpeg lib mentions 

https://github.com/aumuell/libjpeg-turbo/tree/master/java
```
Note for OS X users
-------------------

/usr/lib, the directory under which libturbojpeg.dylib is installed on Mac
systems, is not part of the normal Java library path.  Thus, when running a
Java application that uses TurboJPEG/OSS on Mac systems, you will need to pass
an argument of -Djava.library.path=/usr/lib to java.

```

I'll try to make it local, easier for the lib.

ThomasLengeling, 2015-07-03:
fix with c454e77f38f459271a5fa2ec781f7d48f9c5656c

ThomasLengeling, 2015-07-06:
I just tested the openkinect2 library on fresh mac, you only need to install the turbo jpeg, its just a simple .pkg installation process.

http://sourceforge.net/projects/libjpeg-turbo/

After that just copy the openKinect2 folder to processing library folder, and it works!!.

shiffman, 2015-07-07:
I've got it working and sent you an e-mail with more details.  Progress is in https://github.com/shiffman/OpenKinect-for-Processing/tree/processing-3.  Will leave this open as I'd like to remove the external dependency eventually.  Perhaps this solution can also apply here.

https://github.com/shiffman/OpenKinect-for-Processing/wiki/install_name_tool

Feel free to reply to my e-mail by opening new issues on github if you prefer!

ThomasLengeling, 2015-07-14:
great just made the changes to libJNILibfreenect2.lib. loads libturbojpeg.dylib locally with

```install_name_tool -change /opt/libjpeg-turbo/lib/libturbojpeg.0.dylib  @loader_path/libturbojpeg.dylib  libJNILibfreenect2.dylib```

shiffman, 2015-07-24:
Based on my testing I believe your latest change resolved this!

closed 2015-07-24 by shiffman