shiffman/OpenKinect-for-Processing #15: Update documentation

shiffman, 2015-07-02:
* [ ] http://shiffman.net/p5/kinect/
* [ ] javadocs?
* [ ] new README
* [ ] clean up example comments and style
* [ ] new examples?

shiffman, 2015-07-06:
* [ ] Tutorial needs new images / videos

shiffman, 2015-07-07:
Documentation needs all new stuff for Kinect v2!

shiffman, 2015-08-20:
I did a pass over the documentation page  http://shiffman.net/p5/kinect/ but I will leave this open as I'd like to make a video tutorial and other improvements still.

johnbcarpenter, 2016-06-28:
@shiffman ... Kinect CinderBlock is 3 years out of date... not sure it's being maintained any more or is still a helpful reference.  i spent some time trying to get the examples to build and ran into some known (old) issues last night... finally gave up.  

another note on > Kinect 1473: This looks identical to the 1414, but is an updated model. It should work with this library, but I don’t have one to test. Please let me know if it does or does not!

1473 definitely works, though i see open issue #45