shiffman/OpenKinect-for-Processing #21: kinect v2 raw depth data

shiffman, 2015-07-08:
The v1 kinect raw depth data range is 0 - 2048.  

* What is the raw depth data range for kinect v2?
* How do the v2 depth values map to distance for thresholding and point clouds?

The answers to these questions will provide the fixes for:

* [average point tracking v2](https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v2/AveragePointTracking2)
* [depth threshold v2](https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v2/DepthThreshold2)
* [point cloud v2](https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v2/PointCloud2)

Velro, 2015-07-08:
Dimensions are 512 by 424, with depth values ranging from 0 to 8000mm

ThomasLengeling, 2015-07-08:
I don't think the depth detects upto 8m. 

The ```getRawDepthData``` function return an array of ints from 0 - 4500mm

ThomasLengeling, 2015-07-08:
The ```getRawDepthData``` is throwing  strange values, fixing.

ThomasLengeling, 2015-07-08:
Fixed with f4f1c14e1bc642177a36069ac1a14a9a6e296795

shiffman, 2015-07-08:
Awesome, going to close this issue and we can open new ones if there are more specific bugs with particular examples, etc.

closed 2015-07-08 by shiffman

lisajamhoury, 2018-07-18:
For other people searching for this info. This is from the Kinect [SDK](https://docs.microsoft.com/en-us/previous-versions/windows/kinect/dn772983(v%3dieb.10)): 

> The data for this frame is stored as 16-bit unsigned integers, where each value represents the distance in millimeters. The maximum depth distance is 8 meters, although reliability starts to degrade at around 4.5 meters. Developers can use the depth frame to build custom tracking algorithms in cases where the IBodyFrame isn’t enough.

Eight meters would mean 0-8000, or 13-bit, which goes up to 8191, with reliability beginning to degrade at 4500.