shiffman/OpenKinect-for-Processing #22: Enables for Kinect v2

ThomasLengeling, 2015-07-08:
include enables for each capture frame same as kinect v1?

startDepth()
startVideo()
startIR()
startRegistered()

ThomasLengeling, 2015-07-08:
as well as

toggleVideo()
toggleDepth()
toggleIR()

ThomasLengeling, 2015-07-08:
Enables working with 02caf004538433958e8e8c109b81f4dd11847bcb

shiffman, 2015-07-08:
Thanks for this.  I'm not married to the particular functions and their names at all.  I made some very arbitrary decisions.  Back when I made the first Kinect library being able to process only, say, the depth had significant performance benefits.  But testing now, everything seems to just run so fast even with everything going.  

If you have better ideas for how to name / organize the methods I'd love to hear them. (For example, with v1, I find the the `setIR()` and `setColorDepth()` methods a bit awkward.)

ThomasLengeling, 2015-08-03:
change the naming check the branch rectacto_naming, updated all the examples

shiffman, 2015-08-08:
This looks great to me!  What about removing the toggle functions and not having the functionality in the examples where the depth and video images can stopped and started?  This seems like no one will every use this functionality?

ThomasLengeling, 2015-08-08:
sure I can remove the toggle functions

ThomasLengeling, 2015-08-08:
Do I remove the stopDepth(), and stopVideo() functions?

shiffman, 2015-08-08:
I think we should leave the various `stop()` methods but not include them as a key aspect of documentation/examples.

shiffman, 2015-08-20:
All set now, thanks @ThomasLengeling !

closed 2015-08-20 by shiffman