shiffman/OpenKinect-for-Processing #23: Only works properly when the sketch is run as a 32-bit application

iaaaan, 2015-07-11:
Hi everyone,

"A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit  application."
This error is back since 0.3a4. Please note it used to work properly with 0.3a3.

In case that helps, I run the library examples in Processing 3.0a10, on Mac OSX 10.9.5. Please let me know if you need further info.

Also, I just wanted to add a general thank you for your work @shiffman, it's amazing stuff (:

shiffman, 2015-07-11:
I believe this is due to #12.  Can you confirm that it works if you install [Turbo JPEG for Mac](http://sourceforge.net/projects/libjpeg-turbo/)?

shiffman, 2015-07-11:
(Are you using the newest release?) 

https://github.com/shiffman/OpenKinect-for-Processing/releases

iaaaan, 2015-07-11:
Indeed, I'm using the newest release (0.3a4), and installing Turbo JPEG fixes the issue.

closed 2015-07-13 by ThomasLengeling