shiffman/OpenKinect-for-Processing #25: freenect-glview issue

yishiyang, 2015-07-20:
Hi,

I'm a designer trying to use Kinect with processing on OS X Yosemite. The Kinect is model 1520 and it is connected to my laptop though the Windows Adapter. After installed freenect, I tried "freenect-glview" and always got:

Kinect camera test
Number of devices found: 0

However the device is listed in the USB device tree.
[image]

Tried to search online but can't find any solutions. Please help!

yishiyang, 2015-07-20:
Installed libjpeg-turbo. Now I can run some examples, but the console says:  No Device Connected!

ThomasLengeling, 2015-07-20:
Model 1520 is the Kinect one right?.

Try a couple of times restarting the sketch and unplugging the Kinect v2 several times. 
Sometimes the USB doesn't recognize the devices right away.

When it says "No device Connected" unplugged the power or disconnect the usb until it recognize the Kinect v2. It will say 1 Device Connected. After that just restart the sketch if you get no image data.

thomas

yishiyang, 2015-07-20:
It works! I don't what happened... but after several unplugging and restarting sketches it finally worked!

Thanks a lot!

One more questions, for the "PointCloud2" example, it says "the function getRawDepthData() does not exist." How can I solve this?

ThomasLengeling, 2015-07-20:
amm, yeah the release files are not up-to date with the github code. You could download the repository, and copy the the libs inside the lib folder. so openKinect-for-Processing/lib to YourProcessingSketches/libraries/openKinect-for-Processing/library and do the same for the examples in the repository.
There are two new very cool Point Cloud examples.
Tell if that works for you

yishiyang, 2015-07-20:
Did what you said, got an error "The function startDepth() does not exist."

ThomasLengeling, 2015-07-20:
ok, Fixed.

Try the new release

https://github.com/shiffman/OpenKinect-for-Processing/releases/tag/0.3a5

shiffman, 2015-07-24:
Closing, please re-open if the issue persists.

closed 2015-07-24 by shiffman

yishiyang, 2015-07-26:
Thank you Thomas! The example works! Sorry didn't get time to try during the week.

jeongcatfish, 2018-09-22:
Can I use kinect library on windows 10?
I'm working with kinect v2 on windwos 10 but It doesn't work.. error keep saying "No Device Connected!" "Cannot Find Devices"