shiffman/OpenKinect-for-Processing #26: UnsatisfiedLinkError: no msvc/turbojpeg in java.library.path

paolofuse, 2015-07-21:
I'm using version 0.3a5 on Windows 8.1 64 bit machine, creating simple project in Eclipse.
I have this error:

UnsatisfiedLinkError: no msvc/turbojpeg in java.library.path

Any suggestions?
Thanks

ThomasLengeling, 2015-07-21:
are you using a kinect v2?

ThomasLengeling, 2015-07-21:
could you try running the latest release  inside the processing IDE.

https://github.com/shiffman/OpenKinect-for-Processing/releases/tag/0.3a5

If you running on eclipse you need to add native path folder location with is the lib folder

paolofuse, 2015-07-22:
Yes I'm using Kinect v2 and yes I've added this library to native path.
I'll try inside Processing IDE but I need to use into Eclipse for projects more complex than a simple sketch.
Thank you

shiffman, 2015-07-24:
Based on my testing I believe this is resolved, we can re-open an issue if it comes up again.  Thanks @ThomasLengeling !!!

closed 2015-07-24 by shiffman

RATIU5, 2020-11-02:
This issue is old, but I am coming across this same problem in this repo. My project will work fine with this: https://github.com/ThomasLengeling/KinectPV2, no problems. When I moved to this repo, I get `java.lang.UnsatisfiedLinkError: no v2/msvc/turbojpeg in java.library.path`.

So I am using IntelliJ IDEA.  My code looks something like this:
```
import org.openkinect.processing.*;
import processing.core.*;

public class App extends PApplet {
    Kinect2 kinect;

    @Override
    public void settings() {
        size(800, 600, P3D);
    }

    @Override
    public void setup() {
        kinect = new Kinect2(this);
        kinect.initDepth();
        kinect.init();
    }

    @Override
    public void draw() {
        background(0);
        // ...
    }

    public static void main(String[] args) {
        PApplet.main("App");
    }
}
```
I get the error with `kinect = new Kinect2(this);`. For this project, I have added the path to this location: https://i.imgur.com/pASZf9u.png (Content of folder includes `turbojpeg.dll`, `turbojpeg.lib`, among others) in the project's libraries in IDEA, as global library includes. I even tried adding this argument: `-Djava.library.path="my\path\to\folder"`. Not sure why this is happening, it works fine in the Processing IDE but I would rather use IntelliJ IDEA.