shiffman/OpenKinect-for-Processing #27: Optimize depthToWorld calculation

ThomasLengeling, 2015-08-03:
Optimized rawDepth with 4508ff9521108b63934c2cf5c0a53e8946b2caac

Recommendation: evaluate the depth to world positions ( [point cloud example](https://github.com/shiffman/OpenKinect-for-Processing/blob/master/OpenKinect-Processing/examples/Kinect_v1/PointCloud/PointCloud.pde) ) it the [data method](https://github.com/shiffman/OpenKinect-for-Processing/blob/master/OpenKinect-Processing/src/org/openkinect/processing/DepthImage.java).

```
PVector depthToWorld(int x, int y, int depthValue) {

  final double fx_d = 1.0 / 5.9421434211923247e+02;
  final double fy_d = 1.0 / 5.9104053696870778e+02;
  final double cx_d = 3.3930780975300314e+02;
  final double cy_d = 2.4273913761751615e+02;

  PVector result = new PVector();
  double depth =  depthLookUp[depthValue];//rawDepthToMeters(depthValue);
  result.x = (float)((x - cx_d) * depth * fx_d);
  result.y = (float)((y - cy_d) * depth * fy_d);
  result.z = (float)(depth);
  return result;
}
```

If we do this we should be able to obtain a FloatBuffer with all the depth world positions, which would be great for advance users that want to process the complete point cloud using openGL.

Doing this we could have two examples; the current Point cloud example that calculates the depth to world positions and another one using a built-in function that evaluates the depth to world positions.

shiffman, 2015-08-08:
Yes, I think this is a good idea!

ThomasLengeling, 2015-08-13:
ok, the changes are done with 
https://github.com/shiffman/OpenKinect-for-Processing/releases/tag/0.3a6

shiffman, 2015-08-14:
Awesome, thanks!

closed 2015-08-14 by shiffman