shiffman/OpenKinect-for-Processing #31: Get release into contributions manager for Processing 3

shiffman, 2015-08-08:


tburcham, 2015-12-04:
Hey, is this a Processing 3 - specific error?  I've not been able to get anything running, even going through some of the old 'install libusb' notes I've found.

ThomasLengeling, 2015-12-04:
what is your set up?, OS, Kinect v1 or v2, error console?

tburcham, 2015-12-04:
So sorry!

Osx el capitan
Kinectv2
Processing 3

Will need to grab the actual error later, it was similar to the 'dependencies required' error on issue #30, I believe.

Tim

ThomasLengeling, 2015-12-04:
amm try connecting and disconnecting the Kinect v2, some times it does not detect the device right away and you have to unplug it or restart the sketch. I am going to update the core lib soon, hope that helps

tburcham, 2015-12-08:
Here's the error:
java.lang.UnsatisfiedLinkError: Can't load library: /Users/timburcham/Documents/Processing/libraries/openkinect_processing/library/v2/mac/libturbojpeg.dylib
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:441)
	at java.lang.Thread.run(Thread.java:745)
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.

Here's the code:

import org.openkinect.freenect.*;
import org.openkinect.freenect2.*;
import org.openkinect.processing.*;
import org.openkinect.tests.*;

Kinect2 kinect2;

void setup() {
  size(1024, 848, P2D);

  kinect2 = new Kinect2(this);
  kinect2.initVideo();
  kinect2.initDepth();
  kinect2.initIR();
  kinect2.initRegistered();
  // Start all data
  kinect2.initDevice();
}

void draw() {
  background(0);
  image(kinect2.getVideoImage(), 0, 0, kinect2.colorWidth*0.267, kinect2.colorHeight*0.267);
  image(kinect2.getDepthImage(), kinect2.depthWidth, 0);
  image(kinect2.getIrImage(), 0, kinect2.depthHeight);

  image(kinect2.getRegisteredImage(), kinect2.depthWidth, kinect2.depthHeight);
  fill(255);
  text("Framerate: " + (int)(frameRate), 10, 515);
}

tburcham, 2015-12-30:
I was able to get everything working with a full reinstall.  Thank you, this lib will get me off of my Windows drive and onto my proper Mac :)

closed 2016-01-05 by shiffman