shiffman/OpenKinect-for-Processing #32: AveragePointTracking2 Example Issue

Lachie9383, 2015-08-11:
I am getting weird behaviour with theAveragePointTracking2 example and seen in the screenshot below.  

[image]

I'm also receiving this error in the console:

[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 896

Any idea what's happening?

ThomasLengeling, 2015-08-12:
this code should work

https://github.com/shiffman/OpenKinect-for-Processing/tree/refacto_naming/OpenKinect-Processing/examples/Kinect_v2/AveragePointTracking2

however, the initDepth(), and initDevice() naming should be for you startDepth and startDevice I believe

ThomasLengeling, 2015-08-13:
hey!.

check the new release, that example should be working

https://github.com/shiffman/OpenKinect-for-Processing/releases/tag/0.3a6

shiffman, 2015-08-14:
Closing, @Lachie9383 please feel free to test the new release.

closed 2015-08-14 by shiffman