shiffman/OpenKinect-for-Processing #33: Ready to merge refacto_naming branch

ThomasLengeling, 2015-08-13:
New release with the refacto_naming branch.
https://github.com/shiffman/OpenKinect-for-Processing/tree/refacto_naming

Release 
https://github.com/shiffman/OpenKinect-for-Processing/releases/tag/0.3a6

changes include

- Kinect v1 and Kinect v2 has windows and mac support.
- Renaming methods from startDepth to initDepth similar for the other frames.
- Faster point cloud evaluation for v1.
- New point cloud examples for v1 and v2.
- All examples tested
- Mirror mode.

shiffman, 2015-08-14:
Yes, go ahead and merge!   I keep holding off on updating the official release since I'd like to finish off the documentation, but I think I should go ahead.  Merge and I'll make a release and alert @prisonerjohn.

ThomasLengeling, 2015-08-14:
ok, got the linux version working for v1!

shiffman, 2015-08-14:
wow, awesome!

ThomasLengeling, 2015-08-14:
but have to do sudo ./processing to get it activated/. any idea how to do a work around it?, i tried doing 
"chmod 777 libfreenect.so" but no luck.

shiffman, 2015-08-14:
Sadly, I have no linux experience here.  I wonder if @codeanticode might have any tips?

shiffman, 2015-08-20:
I am closing this, a separate issue can be opened to track linux.

closed 2015-08-20 by shiffman