shiffman/OpenKinect-for-Processing #34: There are no kinects, returning null

PotatoSauceMC, 2015-08-14:
I have the Windows Kinect 1.8 Sdk installed and the Kinect Works fine in other software but when I try to run the RGBDepthTest in processing I get a error: 
"64 windows
There are no kinects, returning null"

Then after I close the program I get: 
"Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled."

ThomasLengeling, 2015-08-14:
have you installed the libusbK drivers using libusbK?

https://github.com/shiffman/OpenKinect-for-Processing/tree/refacto_naming#kinect-v1-for-windows

change it for the camara NUI, motor and Kinect NUI

ThomasLengeling, 2015-08-14:
I mean using Zadig  software

PotatoSauceMC, 2015-08-14:
Now its working thank you.

closed 2015-08-14 by PotatoSauceMC

ThomasLengeling, 2015-08-14:
great!!, thanks for testing that out!

ajwells, 2015-11-02:
I am having the same problem. I could not find the Kinect NUI device, I only had camera and motor NUI (as well as Kinect for Windows Security and Kinect USB Audio).

PotatoSauceMC, 2015-11-02:
@ajwells Are you using the Kinect from the Xbox 360? (Kinect v1)

ajwells, 2015-11-02:
Yes the 1414 version

ajwells, 2015-11-03:
I got it working, I was missing the Kinect USB Audio (Composite Parent) as it hides them by default in Zadig

maria4lexy, 2016-10-02:
i'm having the same problem and it's still not working...please help

nidhzam, 2016-10-11:
im having the same problem even after install the Zadig driver

juanigp, 2016-10-30:
@nidhzam I went to device manager, uninstalled the previous drivers I had and then went to zadig, options, list all devices, and installed the libusbK drivers for all my kinect devices showing. Maybe you have already done this, but the first time I got it wrong

candanerim, 2017-02-13:
Hi, I met the same error "Isochronous transfer error: 1" when I tried to run Shiffman's [https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v1/AveragePointTracking] example code for Kinect. I have Kinect v1 1473 and Windows 10 and downloaded drivers with Zadig. I searched the forums, there are so many people who met this error but still couldn't fix. If you find a solution please let me know.

nidhzam, 2017-02-14:
@candanerim did u try to install libusbK drivers.

yashredevil, 2017-04-16:
I have changed Xbox NUI camera and Kinect USB audio to libusbK. Also try unchecking ignore hubs or composite pair option in Zadig and do the same for all Kinect devices . Solved the problem for me. Im still getting "Isochronous transfer error: 1" though, but the sensor is working.