shiffman/OpenKinect-for-Processing #35: linux support

shiffman, 2015-08-20:
from #33, linux is working but

> have to do sudo ./processing to get it activated/. any idea how to do a work around it?, i tried doing 
"chmod 777 libfreenect.so" but no luck.

danbernier, 2015-11-11:
I'm having this issue, too, and `sudo ./processing` fixed it.

I have an XBox 360 model 1473, running on Mint/Cinnamon. Let me know if you need help testing, etc. I'm willing to help with code/bug fixes where I can, but I feel out of my hack-depth here :smile:

ThomasLengeling, 2015-11-15:
Hello @danbernier, do you by chance have a MAC or a PC?, if you have one of them could you try out the lib, it seems that with that Kinect model the library is not working properly. I only have the model 1414
thomas

danbernier, 2015-11-16:
I don't, no. I'll check around to see if I can borrow a Mac, but I wouldn't count on it.