shiffman/OpenKinect-for-Processing #38: libusb-error

Skofo1, 2015-08-26:
First of all thanks for the great work (and while I'm at it, congratulations on the best programming book I've ever read which is Nature of Code). I'm experiencing some issues on my Macbook Pro Retina Late 2013 with OS X Yosemite, Kinect 2. When running any Code, I get these Error messages:

64 mac
:/Applications/Processing.app/Contents/Java/core/library:/Users/moritz/Dropbox/Sketches/libraries/openkinect_processing/library:/Applications/Processing.app/Contents/MacOS/v2/mac
Found path: /Users/moritz/Dropbox/Sketches/libraries/openkinect_processing/library/v2/mac/
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 6 usb devices connected
[Freenect2Impl] found valid Kinect v2 @20:23 with serial 038711145147
[Freenect2Impl] found 1 devices
1 Device Connected!
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) Core(TM) i5-4258U CPU @ 2.40GHz (CPU)[Intel]
  1: Iris (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Iris (GPU)[Intel]
Devce: 0
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 6 usb devices connected
[Freenect2Impl] found valid Kinect v2 @20:23 with serial 038711145147
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[UsbControl::claimInterfaces(IrInterfaceId)] failed! libusb error -3: LIBUSB_ERROR_ACCESS
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @20:23!
no device connected or failure opening the default one!

Any idea on how to fix this? I seem to have bad luck when it comes to libusb...

Cheers
Moritz

shiffman, 2015-08-27:
I'm a little stumped on this one, but will be doing some more testing soon and see if I can reproduce.

Skofo1, 2015-08-27:
tried running it as admin, didn't work neither. I'll try it with another Kinect 2 on another USB Port. Can I help you with anything else? 

Just to make sure: There is no need to install anything else apart from the library on Mac, right?

shiffman, 2015-08-27:
That's correct, the library should work with no additional installs.

ThomasLengeling, 2015-08-27:
I'll give it a check today

ThomasLengeling, 2015-08-27:
I just tried it with the new processing b5 and it works fine for me. With an Intel HD  Graphics 4000

Make sure that you don't have anything else connected in the USB Hubs.
LIBUSB_ERROR_ACCESS error is probably because there is not enough bandwidth to transfer data from the Kinect to the USB port. 
Try it several times, in occasions it takes a couple of tries to get it running.

Skofo1, 2015-08-28:
Interesting - I've just tried it again:

64 mac
:/Applications/Processing.app/Contents/Java/core/library:/Users/moritz/Dropbox/Sketches/libraries/openkinect_processing/library:/Applications/Processing.app/Contents/MacOS/v2/mac
Found path: /Users/moritz/Dropbox/Sketches/libraries/openkinect_processing/library/v2/mac/
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 6 usb devices connected
[Freenect2Impl] found valid Kinect v2 @20:6 with serial 254722633847
[Freenect2Impl] found 1 devices
1 Device Connected!
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) Core(TM) i5-4258U CPU @ 2.40GHz (CPU)[Intel]
  1: Iris (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Iris (GPU)[Intel]
Devce: 0
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 6 usb devices connected
[Freenect2Impl] found valid Kinect v2 @20:6 with serial 254722633847
[Freenect2Impl] found 1 devices
[Freenect2Impl] failed to reset Kinect v2 @20:6!
no device connected or failure opening the default one!

As you can see, the libusb error is gone, however the kinect is still not working. Tried it with 2 Kinects and both USB Ports, no another USB Devices connected..

ThomasLengeling, 2015-08-28:
Make sure that the power plug is color white. Try several times unplug it and connecting the USB and power source.

Skofo1, 2015-08-31:
Worked funnily enough after unplugging the Kinect power cord several times. Thank you! Can you tell me a little more as to why this is happening?

ThomasLengeling, 2015-09-01:
I have no idea, sometimes it occurs with libfreenect2, however, in processing it occurs a some what more often.

closed 2015-10-17 by Skofo1

nicolasraga1, 2016-04-22:
Have you figured it out?
I have the same problem

Yaffa1607, 2017-05-01:
I still have the same problem. did anyone find the solution?

lcfsc, 2019-08-01:
I have been banging my head against this problem for weeks out of sheer stubbornness. (Thomas Sanchez Lengeling's Kinect V2 for Processing is basically the same thing and much easier to get up and running, but I couldn't figure out how to get a color image with satisfactorily registered depth values).  Anyway, I think I finally solved it on a brand new installation of Windows 10 x64. 

In Zadig, I checked the box to show all devices and tried installing libusbk for each (and both) of the kinect devices it listed, but I kept getting the same error ([UsbControl::claimInterfaces(IrInterfaceId)] failed! libusb error -3: LIBUSB_ERROR_ACCESS, etc.). Anyway, I finally tried installing libusbk for the Kinect hub, and that solved it! You will have to check the box that says "Ignore hubs or Composite Parents" and something like Xbox NUI Sensor will appear. I replaced that driver with libusbk and now the library works like a charm. 

Other stuff I did previously that didn't fix the problem, but may have contributed to my eventual success: 

- Installed Kinect SDK 1.8
- Installed Kinect Developer Toolkit 1.8
- Copied the entirety of libfreenect2-0.1.0-vs2015-x64\bin into Processing\libraries\openkinect_processing\library\v2\msvc  (replacing the original .dlls)

I hope this helps someone!