shiffman/OpenKinect-for-Processing #41: Can not detect Kinect Model 1414 on Windows 8.1

XOIIO, 2015-10-09:
I recall using this quite some time ago on windows 8.1, however since the update, the files I had stopped working, and with a fresh install of processing, and downloads of the library it refuses to detect my kinect. I am trying on processing 2.2.1 and on 3, but it fails to detect the kinect. Is there an updated driver I need which I missed? I scoured the page and couldn't see any mention of it.

ThomasLengeling, 2015-10-09:
for windows you need to install the USB drivers that are mention in the requirements

XOIIO, 2015-10-09:
Tried that, and replaced the drivers for the everything that seems related to the kinect (on it's port), I don't have a WCID on any of those, and not sure if it is supposed to show up all as one or not,
[image]

XOIIO, 2015-10-10:
Made some progress after fiddling back and forth, there was a kinect management service running, after I got rid of that and removed and re-plugged the kinect it works, for the motor at least, however I can't get any video feeds, and it doesn't list them in zadig either. Seems that if you had the sdk running that can cause an issue perhaps though

XOIIO, 2015-10-10:
Alright so, got camera working with it, however when switching back to the windows driver, the developer sdk can't see it anymore, interesting.

ThomasLengeling, 2015-10-10:
great, to switch back to the SDK, you need to do

```
To uninstall the libusbK driver (and get back the official SDK driver, if installed):

Open Device Manager
Under libusbK USB Devices, right click the "Xbox NUI Sensor (Composite Parent)" device and select uninstall.
Important: Check the "Delete the driver software for this device." checkbox, then click OK.
```

XOIIO, 2015-10-10:
Tried nuking out the drivers and reinstalling them from the SDK several times, I got kinect for windows security to pop back up as well as "audios" in computer management but the sdk refuses to see it still, as well as other software. They showed up in zadig a couple times but now do not again
[image]

Going to re-download the whole sdk to try that.

XOIIO, 2015-10-10:
Still no luck, and I can't see "kinect for windows security" anymore, I can get it working with processing just fine but I can't get anything using the windows sdk to recognize it.

XOIIO, 2015-10-10:
Still refusing to work, I don't get it. Maybe it's because the driver source is libusb? Problem is I have tired uninstalling all the drivers, uninstalling the kinect for windows drivers from control panel, and then reinstalled them with the sdk from microsoft but they still have this version, I recall the ones listed in zadig were newer than what I had, but I've nuked everything out and the damn thing still won't work.

I wish I had checked these before so I could compare them and see if there's some sort of issue. I can't find the kinect drivers anywhere in system32 to make sure they are all gone, so I don't know what the heck I can do.
[image]

XOIIO, 2015-10-10:
The only thing I can think of is if it's because of the drivers being from libusb, but I can't get rid of them no matter what I try, and I can't find where they are actually installed to, so maybe that would do it if someone can point me there.

XOIIO, 2015-10-10:
Booted up my old laptop and found that with just the sdk it seems all the kinect devices are under "kinect for windows", not usb devices, it must be the libusb drivers causing this, but I can't find out how to get rid of them. I mean I nuked out anything saying kinect or libusb, so what the heck :/
[image]

XOIIO, 2015-10-10:
Starting to go through the registry, some stuff was pointing to a folder called usb_driver under my username directory, deleted that but it still didn't work, and for some reason I can't delete these registry entries.Uuuuhg

XOIIO, 2015-10-10:
God, finally, I got a headache over this. I had to use rootkit tools and runassystem to delete registry keys for this, which didn't do the trick, I also tracked down the libusb ones, but then "kinect for windows camera" wouldn't work, apparently that one got wiped out too, so I copied that registry value after finally tracking the damn thing down, edited it to match the others from my main computer which didn't work, so I had to nuke out even more kinect entries, and then try and find a registry fix tool which actually worked with windows 8, since windows 8 doesn't have scanreg anymore.

Finally got the f**ing thing working though, god it was such a massive hassle.

closed 2015-11-02 by ThomasLengeling

escardface13, 2021-03-02:
I have the same problem, can I have some help ? I don't know what to do. Already nuked all the data of Kinect on Windows 8.1 and reinstall the drivers but the SDK still not reading the Kinect