shiffman/OpenKinect-for-Processing #42: OGL stuff not working at all

prismspecs, 2015-10-12:
I've tried it on three computers (one Windows) so it can't just be me, right? This line:

pgl.vertexAttribPointer(vertLoc, 3, PGL.FLOAT, false, 0, depthPositions)

throws an error that says that vertexAttribPointer wants (int,int,int,boolean,int,int) and it seems like depthPositions is a float buffer.

ThomasLengeling, 2015-11-15:
try again the lib, I updated the openGL examples

AndreasRef, 2016-03-01:
I get the same error as prismspecs (Mac OSX 10.11, Processing 3.0.2).

ThomasLengeling, 2016-03-01:
Try the Github version, I updated them for processing 3.0, I don't know if something change in 3.0.2

peebeebee, 2016-04-24:
DepthPointCloud2 still fails on the PGL rendermode (processing 3.0.2, OSX)

bontempos, 2016-07-18:
Hello, anyone, any success running PointCloudOGL for kinect 1?

I couldn't actually make the current library to work on my mac, so I am using the "official" one available at processing's page. In that version, I have an issue at line 60
`pgl.vertexAttribPointer(vertLoc, 3, PGL.FLOAT, false, 0, depthPositions);`

since I am not familiar with shaders at all, I am having a bad time trying to figure out something that works

mcpoieses, 2016-08-02:
Hey, I'm using Kinect 2 in mac OS X with Processing 3.0
Everything are working fine unless when I try to render with shader (PGL :( It gives me an error that I can't understand:

> _Cannot compile fragment shader; ERROR: 0:11; Regular non-array variable ‘fragColor' may nor be redeclared._

Is it possible that the .glsl code must be some different?
[https://github.com/bontempos/processing/tree/master/PointCloudOGL/data](https://github.com/bontempos/processing/tree/master/PointCloudOGL/data)
I hope you can help me!