shiffman/OpenKinect-for-Processing #43: Kinect2 PointCloud Examples Issues

philspitlerSF, 2015-10-18:
I'm getting an error in the Kinect2 Pointcloud examples.

It errors on this line.

pgl.vertexAttribPointer(vertLoc, 3, PGL.FLOAT, false, 0, depthPositions);

The error is:

the function ("vertexAtrribPointer()" expects parameters like: ""vertexAtrribPointer(int,int,int,boolean,int,int)"

I'm using Processing 3.0

Any ideas?

Thanks

Phil

ThomasLengeling, 2015-11-15:
Try the lib again, the only example that is not showing correctly the color is the colorPointCloud2.

chazevans, 2016-06-29:
I am getting the same error for the ColorPointCloud2 + DepthPointCloud2 examples:

The function "vertexAttribPointer()" expects parameters like: "vertexAttribPointer(int, int, int, boolean, int, int)"

Using:
1.0 release of the openkinect_processing library from Github
Processing 3.1.1
Mac OSX 10.11.5

Thanks in advance for any help!

metaColin, 2016-08-30:
I am having the same bug. Any solutions yet?

MilanMeetKaur, 2017-03-21:
I am facing the same issue for DepthPointCloud2.
Please suggest a solution for the same.
Thanks!