shiffman/OpenKinect-for-Processing #45: 1473 fails *every other* time

shiffman, 2015-10-30:
For some reason the 1473 doesn't work every other time.  I suspect this has something to do with the kinect not shutting down properly.  The error message is:

```
Unable to claim interface -7
Could not claim interface on audio: -99
Failed to open motor subddevice or it is not disabled.
Failed to open audio subdevice or it is not disabled.
There are no kinects, returning null
```

Amusingly, if you just try again it works.  In other words, with the following code, after it fails I click the mouse and it comes to life.

```java
  void setup() {
    kinect = new Kinect(this);
    kinect.initDepth();
  }

  void mousePressed() {
    kinect = new Kinect(this);
    kinect.initDepth();
 }
```

@ThomasLengeling did you ever try with a 1473?  Do you have one? I've attempted various things to make sure the kinect "shuts down" when the Processing sketch closes but no luck.

I wonder if this is just a 1473 bug and not our library.  I don't see anything about this in https://github.com/OpenKinect/libfreenect/.

ThomasLengeling, 2015-11-02:
I'll give it a look, strange behavior.
I don't think I have a 1473 model, I can try to access one.

ThomasLengeling, 2015-11-02:
I wonder if its relative to this issue #39

savvasio, 2015-12-13:
Unfortunately it doesn't work for me. I need to run the sketch twice in order for the Kinect to work. Any other way to properly shut down the depth before exiting the app?

AndreasRef, 2016-03-01:
I have the same issue...

NickCrews, 2016-03-02:
This is also happening to me as well, with model 1473. "Unable to claim interface -7
Could not claim interface on audio: -99" errors alternate with the errors "upload_firmware: failed to find firmware file.
upload_firmware failed: -2" on every other run, not sure if they're related.

smokhov, 2016-03-08:
As a data point, I also observed the same behaviour with 1473 on every other time while ago, but never had a chance reporting it, so I guess this is a prevailing reproduceable issue.