shiffman/OpenKinect-for-Processing #48: UnsatisfiedLinkError when exporting as runnable jar from Eclipse

Skofo1, 2015-11-09:
Hey,

first of all thanks for the great work! I've stumbled upon this error when I tried to export my project as runnable jar with the option "Extract required libraries into generated JAR" and haven't been able to solve it so far:

Found path: v2/mac/
Exception in thread "Thread-2" java.lang.UnsatisfiedLinkError: Expecting an absolute path of the library: v2/mac/libturbojpeg.dylib
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:441)
	at java.lang.Thread.run(Thread.java:745)

 Running the project from within eclipse is no problem. I tried copying the v2 folder into the folder of the .jar, but this didn't work as well. I've noticed, that when I use System.getProperty("java.library.path") in eclipse, it gives me the correct path to the openkinect_processing folder. However, if I print the output of same method after I exported it as a JAR it returns "". 

Cheers
Moritz

ThomasLengeling, 2015-11-15:
fixed with 5046fddc23a3d7225a745f8c6e403f3d9e9680a6
I just tried it and it worked for me, let me if its working

thomas

MoritzSkowronski, 2015-11-30:
works like a charm. 

Thanks!

closed 2015-11-30 by MoritzSkowronski