shiffman/OpenKinect-for-Processing #52: Stop running after 30 minutes / 1 hour

paolofuse, 2015-11-27:
Hi,
I'm using the library on my Mac Osx 10.11.1. All goes well but after 30 minutes / 1 hour kinect stops to work and gives me this error:

[TransferPool] failed to submit transfer: LIBUSB_ERROR_OTHER

I see that libFreenect2 has been updated. Is it possible to update the libraries used by OpenKinect for Processing?
On a Window machine kinect works well all the time.

Thanks a lot.

bmv437, 2016-03-17:
I've also run into that LIBUSB error randomly. Did you ever find a solution?

ThomasLengeling, 2016-03-17:
is the same issue for the Kinect 2?

paolofuse, 2016-03-17:
I had the issue just with Kinect 2, I didn't try Kinect 1.
My solution was looking pure libfreenect2 library and then modify the file libfreenect2.cpp.
Here you can find my solution.
[libfreenect2.cpp.zip](https://github.com/shiffman/OpenKinect-for-Processing/files/178225/libfreenect2.cpp.zip)

Hope can help you too.

ThomasLengeling, 2016-03-17:
ok thanks, I will try to update the lib with the new version of libfreenct2, apparently Nite is also working on the new version.

paolofuse, 2016-03-17:
Nite is also working on Mac OSX?

ThomasLengeling, 2016-03-17:
I tried it on Windows, and it worked, but a little buggy

paolofuse, 2016-03-17:
I tried libfreenct2 with OpenNI2 and it worked, but without Nite functions.