shiffman/OpenKinect-for-Processing #54: win10 + kinect 1414 (v1) - "The function kinect.initDevice() does not exist"

JoseIsmaelGraca, 2015-12-06:
When I try to run:

import org.openkinect.freenect.*;
import org.openkinect.freenect2.*;
import org.openkinect.processing.*;
import org.openkinect.tests.*;

Kinect kinect;

void setup() {
  size(512, 424);
  kinect = new Kinect(this);
  
  kinect.initDepth();
  kinect.initDevice();
  
}

void draw() {
  background(0);
}

console error:
 "The function kinect.initDevice() does not exist"

When I run RGBDepthTest.pde processing version 3 shows no response.

Thanks

duelingbanjos, 2015-12-08:
Hi there,

I was having the very same problem - model 1414 + Windows 10

I installed all the newest version of OpenNI - http://structure.io/openni
and also the 1.8 SDK From the microsoft website - the SDK 2.0 is for the Kinect v2

Also it's important to uninstall all the kinect drivers before you do this *I think* & also to have the Kinect unplugged while you're installing the drivers etc. 

Hope this helps !

ThomasLengeling, 2015-12-08:
Hello I think you are missing installing the USB driver for the kinect.

https://github.com/shiffman/OpenKinect-for-Processing#windows

The installation can be tricky

keltroth, 2016-02-02:
Hello,

I found something in openkinect for processing source code : 

`kinect.activateDevice(0); `

Worked for me

smokhov, 2016-03-08:
Issue #20 I guess is incomplete on API alignment.

bonice91, 2016-06-08:
Srry my bad english....

I have the same problem....  initDevice(); does not exists

My OS is Windows 8.1 
Kinect v1 Model 1473

I already installed:
-kinect sdk v1.8
-newest version OpenNI

AbdullahAslam, 2017-03-28:
@keltroth Thank you so much mate. 
kinect.activateDevice(0);  worked for me... but still have issue.. it is saying There are no kinects, returning null. And my kinect is connected. what might be the problem? 
Your reply will be appreciated !
Thanks again