shiffman/OpenKinect-for-Processing #55: Kinect array not working: Two Kinects 1473 & One Kinect 1414

Gabirell, 2016-01-14:
Hello!
I'm trying to make a 3 kinect array with two kinect models 1473 and one model 1414 using your "multikinect" code.
When I connect the two 1473s only one appears on screen, and when I connect the 1414, only this last one appears, leaving space for the other two that don't show anything. The console shows 3 kinects detected:
<addr>
Found path: /Users/macbookpro/Documents/Processing/libraries/openkinect_processing/library/v1/mac/
Loaded freenect from /Users/macbookpro/Documents/Processing/libraries/openkinect_processing/library/v1/mac/libfreenect.dylib
number of Kinect v1 devices  3
Unable to claim interface -7
Isochronous transfer error: 1
Could not claim interface on audio: -99
Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.There are no kinects, returning null
Unable to claim interface -7
Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.There are no kinects, returning null
Isochronous transfer error: 1
Isochronous transfer error: 1
Isochronous transfer error: 1
Isochronous transfer error: 1
Isochronous transfer error: 1
<addr/>
Any help is appreciated!
thx!
[image]

ThomasLengeling, 2016-01-15:
Hello

Have you tried a simple kinect example using the 1473 model?, I know that the library has some issues with the 1473 model

Gabirell, 2016-01-21:
Thank you for the question, Thomas. Sorry about my delay.
Yes! I've tried them and the result is this:
I discovered that one of my usb inputs is not working fine... I tried connecting the two 1473 on the other usb inputs and they both worked. But not when I mix them (a 1473 & and a 1414) I couldn't try on another computer yet...

smokhov, 2016-03-08:
@Gabirell you also appear to be running on a Mac; I am not aware of MacBooks having 3 USB hubs (not ports, hubs). Generally, you need to run each Kinect device on a separate USB hub/bus. You can run successfully two of them on a MacBook that has a USB port on both sides of its keyboard but not more than two. On a desktop if you have 3 or more USB hubs/controllers (on the motherboard plus expansion cards), then you can have more than 2 devices running. I successfully mixed Kinect v2, 1414, or 1473 on a MacBook with this library with 1473 having the issue described in #45. We also successfully ran two 1473s on Windows 8, but I don't recall trying mixing.