shiffman/OpenKinect-for-Processing #56: OpenKinect does not work

EngDhammika, 2016-01-21:
I am new to kinect and I am using kinect 1414 model
 on windows 8 ,processing 3
 Every time I run open kinect for processing on windows 8
 Bellow error message is print out on console 
"64 windows
 There are no kinects, returning null
 Could not open camera: -12
 Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open  audio subdevice or it is not disabled."
 
 Still my kinect 1414 is working on windows kinect Studio﻿

ThomasLengeling, 2016-01-22:
for windows you need to install the lubusbZ drivers via zadig

https://github.com/shiffman/OpenKinect-for-Processing#windows

https://github.com/OpenKinect/libfreenect#windows

EngDhammika, 2016-01-23:
i installed  lubusbZ drivers also but my porblem is same

ThomasLengeling, 2016-01-23:
if the kinect is still working on the kinect studio that means that you haven't correctly installed the liibusbZ drivers.

EngDhammika, 2016-01-31:
yes now it is not working with kinect studio but also the still error message is same 
also on drivers list 
only Kinect USB Audio and Xbox Nui Motor appeared as connected drivers

SetupandDraw, 2016-03-21:
I have also issues with Openkinect. Working with kinect 1414 with Processing 3 on MAC OSX.
AveragePointTracking, DepthThreshold working fine. 

MultiKinect (tested with just one device) working with no errors untill I press 'i' key for options, then I get these messages from the console:
[Stream 70] Expected 1748 data bytes, but got 948
[Stream 70] Expected 1748 data bytes, but got 948
[Stream 70] Expected 1132 data bytes, but got 316

PointCloud working randomly; sometimes I got the following error message:
send_cmd: Bad len 0004 != 000c
freenect_fetch_reg_pad_info: send_cmd read -1 bytes (expected 8)
freenect_camera_init(): Failed to fetch registration pad info for device
There are no kinects, returning null

RGBDepthTest often I get several random different error messages. Most of the time the following 2: 
1)
java.lang.IndexOutOfBoundsException
	at java.nio.Buffer.checkIndex(Buffer.java:540)
	at java.nio.DirectByteBuffer.get(DirectByteBuffer.java:253)
	at org.openkinect.processing.RGBImage.data(Unknown Source)
	at org.openkinect.processing.Kinect$2.onFrameReceived(Unknown Source)
	at org.openkinect.freenect.Freenect$NativeDevice$1.callback(Unknown Source)
	at org.openkinect.freenect.Freenect.freenect_process_events(Native Method)
	at org.openkinect.freenect.Freenect.access$500(Unknown Source)
	at org.openkinect.freenect.Freenect$EventThread.run(Unknown Source)
Could not run the sketch (Target VM failed to initialize).
For more information, read revisions.txt and Help → Troubleshooting.
2)
Could not claim interface on camera: -3
Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.There are no kinects, returning null

PointCloudOGL got the following error message:
The method vertexAttribPointer(intm int, boolean, int, int) in the type PGL is not applicable for the arguments (int, int, int, boolean, int, FloatBuffer)

zhannar, 2016-06-02:
Did you ever get this error resolved?

The method vertexAttribPointer(intm int, boolean, int, int) in the type PGL is not applicable for the arguments (int, int, int, boolean, int, FloatBuffer).

I also get it when I try to run this: [DepthPointCloud2](https://github.com/shiffman/OpenKinect-for-Processing/blob/master/OpenKinect-Processing/examples/Kinect_v2/DepthPointCloud2/DepthPointCloud2.pde)

AbdullahAslam, 2017-03-28:
getting same issue .. anyone resolved this ?

jango-fx, 2017-03-31:
> The method vertexAttribPointer(intm int, boolean, int, int) in the type PGL is not applicable for the arguments (int, int, int, boolean, int, FloatBuffer)

is solved by changing `pgl.vertexAttribPointer(vertLoc, 3, PGL.FLOAT, false, 0, depthPositions);` to `pgl.vertexAttribPointer(vertLoc, 3, PGL.FLOAT, false, Float.BYTES * 3, 0 );`