shiffman/OpenKinect-for-Processing #57: open kinect for processing library not working

karinkomkov, 2016-02-28:
i have win 8.1 
the processing opens the kinect device only from library :kinect4winSDK

"there are no kinects" --- this is the eror

ALSO,  maybe you can also tell me why the openNI doesn't work. i tried almost everything...[http://imgur.com/vH4N9j8](url)

^ maybe this pic will explain it

http://imgur.com/vH4N9j8

smokhov, 2016-03-08:
For OpenNI comment -- `SimpleOpenNI` simply does not work in Processing 3 (you appear to be running 3.0.2). You have to use Processing 2.2.1 for it. I had no issues running OK4P on Windows 8.1 in either Processing 2.2.1 or 3.x. You have to make sure you don't run them all simultaneously so they don't try to compete for the device you have -- close all, run one at time, then close it (quit), run another.

shiffman, 2016-03-13:
I'm going to close this as the issues here are for bugs with this specific Kinect library.  For questions about SimpleOpenNI or kinect4winSDK I would suggest http://forum.processing.org.

closed 2016-03-13 by shiffman