shiffman/OpenKinect-for-Processing #59: DepthPointCloud2 error

MrfksIv, 2016-03-19:
When i try to run the DepthPointCloud2 i get the exception "CameraParams cannot be resolved to a variable. 
The problem seems to be at the depthToPointCloudPos method, specifically at:

`point.x = (x - CameraParams.cx) * point.z / CameraParams.fx;
  point.y = (y - CameraParams.cy) * point.z / CameraParams.fy;`

Can anyone help?
Thanks

ThomasLengeling, 2016-04-22:
you need to have the adicional tab/code of CameraParams

closed 2016-04-22 by ThomasLengeling

metaColin, 2016-06-23:
I think this is what you need:

//camera information based on the Kinect v2 hardware
static class CameraParams {
  static float cx = 254.878f;
  static float cy = 205.395f;
  static float fx = 365.456f;
  static float fy = 365.456f;
  static float k1 = 0.0905474;
  static float k2 = -0.26819;
  static float k3 = 0.0950862;
  static float p1 = 0.0;
  static float p2 = 0.0;
}