shiffman/OpenKinect-for-Processing #62: Kinect not detected

kitus669, 2016-03-29:
Hello, and first of all thanks for your help.

I am begining to program with kinect and I have a problem which I cannot solve.

I connected kinectv2 and I have installed all the needed tools, and when I am trying to use kinectv2, when using kinect v2 for Processing  examples kinect is working with no problem, but when I play some of the examples of kinect v2 from Open kinect for processing library, for example: AveragePointTrancking2, i get the following error:

64 windows
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @1:9!
[Freenect2Impl] found 0 devices
No Device Connected!
Cannot Find Devices
EXIT
closing Kinect v2 :

I don't understand why it doesn't detect any connected kinect, because the other library examples detect it and make it work

If you have any idea I would appreaciate the help

vkampouris, 2016-11-25:
Same problem here. Did you find a solution? Is it a drivers matter or something?

OmegaNalphA, 2017-01-31:
Same issue here, doesn't seem to have a resolution

SanniClara, 2020-05-21:
Same problem.