shiffman/OpenKinect-for-Processing #69: Trouble With Intel Compute Stick 2016

jwbrobst, 2016-04-25:
When attempting to use the OpenKinect Library on Intel Compute Stick (W10 32bit) I've been unable to successfully run even the example files. The following error message is from the PointCloud example included in the download:

_32 windows
java.lang.UnsatisfiedLinkError: C:\Users\jwbro\OneDrive\Documents\Processing\libraries\openkinect_processing\library\v1\msvc\pthreadVC2.dll: Can't load AMD 64-bit .dll on a IA 32-bit platform
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 64-bit application.
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:447)
	at java.lang.Thread.run(Thread.java:745)_

It's most likely a 32bit error but maybe there's another .dll to replace the one specified by the error? Let me know if there's any fix for this. Thank you so much.

jwbrobst, 2016-04-25:
If it helps, the error may be related to the first variable declaration of _Kinect kinect;_.

jwbrobst, 2016-04-25:
_A library relies on native code that's not available.
Or only works properly when the sketch is run as a 64-bit application._

Is the only constant error throughout the use of this library. This occurs anytime I use _Kinect kinect;_

kkirklewski, 2020-02-25:
Hey @jwbrobst, did you manage to solve this problem?