shiffman/OpenKinect-for-Processing #70: Issue in exporting processing programs using kinect v1

nevahid, 2016-05-02:
hi everybody,
there seems to be an issue when trying to export a program that uses kinect v1 in processing. in PDE environment everything works fine and the exporting is finished successfully. but when running the exported program, just a gray screen appears.
maybe this is similar to problem for kinect v2 export path.
p.s.: i use processing 3.0.1 build 247, openkinect-for-processing v 1.0. tested it under windows xp and windows 8.  
thanks in advance

nevahid, 2016-05-02:
hi again,
i think it might be related to export path, similar to that of v2 kinect. just a guess though...

ThomasLengeling, 2016-05-03:
Hello, copy the .dll files in the lib folder to the location of the exported .exe

nevahid, 2016-05-04:
hi there Thomas, thanks for response... but no improvement yet... trying to figure it what's the problem.

miguel-aragon, 2016-05-07:
Hi, I have the exact same problem trying to export a sketch using kinect v1 on a mac:

Exception in thread "Animation Thread" java.lang.NoClassDefFoundError: com/sun/jna/Library
...
Caused by: java.lang.ClassNotFoundException: com.sun.jna.Library

I tried adding the jna.jar in the processing IDE. It seems that the Eclipse IDE is able to add the jna.jar file correctly:

https://forum.processing.org/one/topic/gsvideo-proclipsing-eclipse.html

nevahid, 2016-05-09:
hi people, i tried using "try catch" to see where the problem is. the exception used is NoClassDefFoundError and the result in getmessage is com/sun/jna/Library and getcause results com.sun.jna.Library. i searched the exported folder and didn't find jna.jar, but i see that you have added it in your "additional libs" folder and in "open kinect for processing", but yet it is not in my computer!
so it seems the problem is about jna.jar... any idea how to solve it?!

nevahid, 2016-05-09:
there seems to be a new version for processing! and i just mentioned update JNA 4.2! i wonder if this solves the issue!!
besides i just wanted to say that dLibs_freenect is a library that is using jna.jar (and including it in its library folder) might help contributors, though it was designed for previous versions of processing...

nevahid, 2016-05-11:
and finally this issue is solved  too :-) ;-)
the answer was ridiculously easy... we just copied the file JNA.JAR to processing library folder.
and now everything is fine... the problematic jna.jar file must be around so that you can run the exported program...
have fun :-)

miguel-aragon, 2016-05-12:
Hi Nevahid, just confirming that your idea works. Thanks!

shiffman, 2016-05-16:
Glad to hear! Is this a bug that needs to be reported at https://github.com/processing/processing?  Or does something need to be fixed here?

nevahid, 2016-05-16:
i'm not sure if it's a bug or not. the problem is about jna.jar. i think it should be put manually in processing folder, or maybe to be copied with processing installation.

brendanmatkin, 2016-06-06:
Hey @shiffman, I THINK the issue needs to be fixed here. 

 jna.jar is in the additional libs folder here on github, but it is not found in <processing sketchbook>/libraries/openkinect_processing. Maybe the library is not being built properly? 

Or is jna.jar something that is included in processing and is just not being included in the exported app (I'm thinking about this because it works fine when just running the sketch, the problem only occurs during export)? Either way jna.jar is not being copied to the exported project - how does Processing usually handle dependant libraries? 

Workaround: If I download the zip of this project from github, and copy jna.jar from it to <my exported app>/contents/java then the app works perfectly fine ( @nevahid 's workaround didn't work for me).

(I'm now using processing 3.1.1 OS X 10.11.5)

shiffman, 2016-06-07:
As far as I know I think `jna.jar` is supposed to come with Processing though it would be easy enough to add it here.  Maybe there is some sort of exports.txt config file that I haven't set up properly? (have a vague memory of something like this) I will try to investigate soon, and welcome any help, thanks for your patience!

brendanmatkin, 2016-06-09:
I found this: https://github.com/processing/processing/wiki/Library-Basics#structure-of-a-library-folder 

So, although jna.jar is in the openkinect_processing/library folder, it might be that it's not being included because it should be inside the operating system folders. OR you can make an export.txt file to define which files are included when, but it looks like the folder approach is preferred. 

Moving jna.jar into \<mysketchbook\>/libraries/openkinect_processing/library/v1/mac seems to fix the problem for me (exporting an app after this works as expected). 

It might be a problem somewhere else (not sure how the additional_libs stuff is being included), but that looks right to me. 

Maybe I will try a pull request soon, but building java libs is something I don't know how to do yet and don't have time at the moment!

johnbcarpenter, 2016-06-28:
thank you @brendanmatkin ... this solution works for me on os x 10.11 + processing 3:

> Workaround: If I download the zip of this project from github, and copy jna.jar from it to <my exported app>/contents/java, then the app works perfectly fine ( @nevahid 's workaround didn't work for me).

Mallinckrodt, 2016-07-04:
Hmm I still cannot fix my issue when I export. I am on on OS X 10.11.3 + processing 3. When I export for Mac OS, there is only the .dmg and a folder called source which holds the .java and .pde. I have tried copying the jna.jar into either folders but still only get a grey screen when I execute it. Any idea?

brendanmatkin, 2016-07-04:
@mallinckrodt 
When you export there should be a .app in the sketch folder. Right click that and click "show package contents". Then find the 'contents' folder, and within that 'java'. Paste jna.jar into that 'java' folder.

bene-ulli, 2016-12-21:
Hello,

this all is not working for me! I use the SimpleOpenNI library to work with the Kinect 1473. For testing the export I used an example ("User3d") offered by SimpleOpenNI. And because this library is not working with Processing 3, I use the Processing 2.2.1. 

When I export the file and then add the jna.jar inside the "User3d.app/contents/java" or into other folders, I still will receive a gray image. 
Has it maybe something to do with the SimpleOpenNI library? Or with the Kinect 1473?

jarbasjacome, 2017-11-26:
Hi all,
Thank you for the informations.
I had the same problem here exporting an openkinect sketch with Processing 3 - 0262, with Arch Linux.
The exported app only work after I copy jna.jar from /usr/share/processing/lib/jna.jar to the exporting generated folder application.linux64/lib/ AND change the generated run script: it needs to reference jna.jar. So you need to add :$APPDIR/lib/jna.jar to the end, like this:
```

#!/bin/sh
APPDIR=$(readlink -f "$0")
APPDIR=$(dirname "$APPDIR")
$APPDIR/java/bin/java -Xms64m -Xmx2000m -Djna.nosys=true -Djava.ext.dirs="$APPDIR/java/lib/ext" -Djava.library.path="$APPDIR:$APPDIR/lib" -cp "$APPDIR:$APPDIR/lib/zumvi.jar:$APPDIR/lib/core.jar:$APPDIR/lib/jogl-all.jar:$APPDIR/lib/gluegen-rt.jar:$APPDIR/lib/jogl-all-natives-linux-amd64.jar:$APPDIR/lib/gluegen-rt-natives-linux-amd64.jar:$APPDIR/lib/openkinect_processing.jar:$APPDIR/lib/jna.jar" zumvi "$@"
```

ps: zumvi is the name of the exported sketch running an instalation after Zumbi dos Palmares (https://en.wikipedia.org/wiki/Zumbi)