shiffman/OpenKinect-for-Processing #73: Isochronous transfer error: 1

uraeu5, 2016-05-05:
I'm using a kinect v1 1414, installed the SDK and installed the libusbK drivers, but whenever I run any of the example files with the open kinect for processing library I get "Isochronous transfer error: 1"  and the sketch freezes.  Any ideas how to solve this?  Much thanks.

laurage, 2016-05-27:
Same here.
I'm on Windows 10.

laurage, 2016-05-27:
Did you find any fix?

laurage, 2016-05-27:
Update: Every time I was turning the computer off, it stopped working again.

Solution:
The drivers must be desinstalled in the Windows manager
+ Use windows clean disk to clear the remaining traces of the drivers!
Then reinstall the drivers with Zadig.

huangy10, 2016-09-26:
Same problem here, on macOS. But situation is a little different here. When I start the program, the kinect works fine at the beginning. But after running for a while, there is a chance that the sketch stops and the error messages shows up (not always). I cant figure out why.

jonnyscholes, 2017-01-11:
This link is useful to those having this issue that came here via Google: https://groups.google.com/forum/#!topic/openkinect/3z8-4xIUXf8

candanerim, 2017-02-13:
Hi, I met the same error "Isochronous transfer error: 1" when I tried to run Shiffman's [https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v1/AveragePointTracking] example code for Kinect. I have Kinect v1 1473 and Windows 10 and downloaded drivers with Zadig. I searched the forums, there are so many people who met this error but still couldn't fix. If you find a solution please let me know.

jnydutra, 2017-09-19:
Downgrade libusbK worked for me. Use http://zadig.akeo.ie/downloads/zadig_2.0.1.exe on windows.

PR #126 (2017-09-21): Update README.md
Closes #73 and #106
commit 47cb8935 (2017-09-21): Update README.md

Closes #73 and #106
closed 2017-09-21 by shiffman