shiffman/OpenKinect-for-Processing #74: Kinect not working, Native library not found?

spiral6, 2016-05-07:
Running the latest version of Processing (3.0.2) and the latest OpenKinect for Processing (1.0) example Depth thresholding doesn't work.

> 64 windows
>A library relies on native code that's not available.
>Or only works properly when the sketch is run as a 32-bit application.

Just opens a GUI of nothing and breaks instantly.

Running a Kinect v1.

arky, 2016-07-23:
I have been able to replicate this problem on Windows 7 (32bit) machine.

carloscastellanos, 2016-10-13:
I am having same problem. Windows 8.1 (64), Kinect V2, Processing 3.2.1

mikebridge, 2017-01-18:
Same with Processing 3.2.3 both 32 and 64 bit, Kinect v2

AbdullahAslam, 2017-03-28:
any one solved this issue ? please guide if anyone does?

mikebridge, 2017-03-28:
@AbdullahAslam  I couldn't get it to work, but I had some success with [KinectPV2](https://github.com/ThomasLengeling/KinectPV2).

idnavid, 2017-09-08:
Has anyone been able to solve this issue? I'm trying to use Kinect v2 on Windows for the first time.