shiffman/OpenKinect-for-Processing #76: No Kinect Devices Found OSX el capitan

aliceEddie, 2016-06-14:
I'm having an issue with the library. 
I'm using a Kinect Model 1414 and when I'm trying to run a sketch I get the error:

No Kinect devices found.
Failed to open camera subdevice or it is not disabled.Failed to open audio subdevice or it is not disabled.There are no kinects, returning null

Couldn't find any solution for this so far. 
Any help is greatly appreciated!

aliceEddie, 2016-06-14:
Solution was rather easy, the outlet the kinect was plugged into was switched off. 
T

closed 2016-06-14 by aliceEddie

shiffman, 2016-06-20:
glad to hear!

EvelynGiron, 2016-09-02:
Hello @aliceEddie can you explain me step by step this? I have the same problem and i speak spanish I can´t understand. Please I have 2 months trying make works the kinect with El Capitan en Mac, always is the same: No Kinect Devices Found.

aliceEddie, 2016-09-02:
Hi @EvelynGiron 

I'm sorry I don't speak Spanish, I'll explain it in English, just ask if anything is unclear.

my problem was really simple:
I didn't plug the power cable in properly.
The USB was connected but not the power cable.

Can you try the Kinect on a different computer?

Does the Kinect show up in your system report? 
Connect the Kinect to your computer. Click on the apple logo in the top left corner, then on "about this mac", it will give you information about your mac and on the bottom it should say "system report"

[image]

Now look for USB (under Hardware), and you will get a list of all connected USB devices. Does the Kinect show up?

EvelynGiron, 2016-09-03:
THANK YOU @aliceEddie for you explanation, isn´t matter that was in English. The USB shows as XBOX NIU Motor… LED was still blinking. I have  a question: the green light of your adapter is lit ? , I think maybe the adapter is not working. 
This is more difficult than it should be. 

[image]
[image]