shiffman/OpenKinect-for-Processing #77: Kinect2 Mac Program Not Exporting

Sensebellum, 2016-06-23:
Lovely library. Many thanks. Feels like the good ol days of the Kinect v1

Anyways, getting started and the code runs great. Although, I did a test export and the code did not run in the exported app. Should this be expected?

Looking to make some little apps and stuff for sharing.

Thank you!

//

import org.openkinect.processing.*;

Kinect2 kinect2;

void setup() {
  size(512, 422);
  frameRate(30);

  kinect2 = new Kinect2(this);
  
  kinect2.initIR();

  kinect2.initDevice();
}

void draw() {
  background (0);

  PImage ir = kinect2.getIrImage();
  image(ir, 0, 0);
}

Sensebellum, 2016-07-13:
Just wanting to ping this one real quick.

Is this issue something to be expected with bleeding edge libraries? Or, have we perhaps committed an error in code or IDE setup/export?

jaysonh, 2017-01-13:
having the same problem, but with kinect v1?

Runs fine from Processing IDE but when I try exporting the app it stalls as soon as I try using the Kinect object?

I'm using Processing 3.1.1 and the latest version of the OpenKinect library.