shiffman/OpenKinect-for-Processing #78: CameraParams does not exist Kinect v2 Mac OS X

Sensebellum, 2016-06-23:
Testing out the depth cloud for the Kinect v2 on the Mac OS X

I have found an error that reads as so.

Is it a typo perhaps? I made sure to upgrade to the latest version of Processing just now.

Many thanks once again,

[image]

ThomasLengeling, 2016-06-24:
you need to download the full library, or the sketch, you are missing the tab/PDE that includes that variable

Sensebellum, 2016-06-27:
I will look into that and give it a shot this evening.

abigpotostew, 2016-08-23:
I had the same issue. The kinect guide on Shiffman's site is a bit misleading; it links to the main pde file rather than the project folder. the link should point [here instead](https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v2/DepthPointCloud2).

shiffman, 2016-08-30:
Are you referring to a link at:

http://shiffman.net/p5/kinect/

Let me know which one and I'll fix!  (I welcome pull requests also here: https://github.com/shiffman/shiffman.net/blob/gh-pages/p5/kinect/index.md

timohausmann, 2018-05-29:
Hey @shiffman , I think they were talking about the Link Code for v2: PointCloud at http://shiffman.net/p5/kinect/. It should point to the [folder](https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v2/DepthPointCloud2) and not the PDE, since it requires the CameraParams and the data folder.

o-pperator, 2019-05-03:
Hey im getting a different error on the same line "CameraParams cannot be resolved to a variable" pretty new to this, not sure where to start looking for a solution. cheers

timohausmann, 2019-05-03:
@o-pperator the variable CameraParams is defined in another file, make sure you have the entire project (with data folder and CameraParams.pde) in your sketch folder and not only DepthPointCloud2.pde (see folder link above).

o-pperator, 2019-05-06:
Thanks, I included the whole project (with the .glsl files) in the sketch folder, and now its coming up "Not expecting symbol "i". which is LATIN SMALL LETTER I." Though there doesnt appear to be any stray i's floating around