shiffman/OpenKinect-for-Processing #79: Error for DepthThreshold2 example

chazevans, 2016-06-29:
Getting this error when running DepthThreshold2 example:

The function "startDepth()" does not exist

on line 28:
kinect2.startDepth();

Using:
1.0 release
Processing 3.1.1
Mac OSX 10.11.5.
Kinect v2 model 1520

Thanks in advance for any advice!

ANyanCatFan, 2016-07-04:
It worked for me when I changed it to ```kinect2.initDepth();```

chazevans, 2016-07-07:
Thanks for the response!

I changed line 28 to kinect2.initDepth(); and a new error appeared:
"The function start() does not exist" on line 29

I tried changing line 29 to kinect2.init();

This got the example to compile without an error message but no depth image appeared when the sketch ran. Any other thoughts or advice you might have is very appreciated. Thanks again.

SandroMiccoli, 2016-07-22:
Ok, I had the same problem. The way it worked was changing the startDepth() for initDepth() and also adding a initDevice() afterwards. Like this:
```
  kinect2 = new Kinect2(this);
  kinect2.initDepth();
  kinect2.initDevice();
```

chazevans, 2016-08-10:
Cool! That did the trick. Thanks so much!

shiffman, 2016-08-12:
i'm re-opening this as the examples should be fixed if they are not working and also perhaps the API re-examined if there is confusion.

johnbcarpenter, 2019-05-15:
just running through some of the V2 examples and noticed this bug in the example still, here's what i changed my setup to -- exactly the same as the above fix: 

`void setup() {
  size(1280, 480);

  kinect2 = new Kinect2(this);
  kinect2.initDepth(); // startDepth(); <-- V1
  // Start all data
  kinect2.initDevice(); // needed for V2, not start();

  // Blank image
  depthImg = new PImage(kinect2.depthWidth, kinect2.depthHeight);
}`