shiffman/OpenKinect-for-Processing #80: No Device Connected!

1andres1, 2016-07-20:
64 mac
:/Applications/Processing.app/Contents/Java/core/library:/Volumes/FGHMacPro/Documents/libraries/openkinect_processing/library:/Applications/Processing.app/Contents/MacOS/v2/mac
Found path: /Volumes/FGHMacPro/Documents/libraries/openkinect_processing/library/v2/mac/
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found 0 devices
No Device Connected!
Cannot Find Devices

My computer doesn't recognizes the Kinect Device, I'm working with a Kinect V2, OSX El Capitan 10.11.6, Processing 3. (I have usb 3.0) 

**With Yosemite**
[image]
[image]

**with el capitan**

GaoYingGRE, 2016-09-26:
same issue.

carloscastellanos, 2016-10-14:
same issue, with yosemite

eratatat, 2016-12-21:
Please can someone provide a solution to this. I'm having the same issue on mavericks.

jeongcatfish, 2018-09-22:
I have same issue.
Did you solve it?