shiffman/OpenKinect-for-Processing #82: The function initDevice() does not exist

dtheva, 2016-07-30:
Posting this issue as well as possible resolution.

Running Processing 3.1.1 on fresh install of OS X El Capitan (10.11.6)
Kinect V1 

The following code was obtained from: [Youtube Link from Daniel Shiffman's Coding Rainbow channel](https://www.youtube.com/watch?v=FBmxc4EyVjs)
``` java
import org.openkinect.processing.*;

Kinect kinect; 

void setup()
{
   size(640,480);
   kinect = new Kinect(this);
   
   kinect.initDepth();
   kinect.initDevice();
}

void draw()
{
   background(0);
   PImage img = kinect.getDepthImage();
   image(img, 0, 0);
}
```

If you are using a KinectV1 device, the line:
`kinect.initDevice();`
will throw a "The function "initDevice()" does not exist" error.

I suspect this is due to the fact that initDevice() starts up the IR, depth, and video acquisition. Since KinectV1 cannot receive both video and IR simultaneous (something KinectV2 is able to do however), this function call is not applicable to the KinectV1 device.  The line can be omitted (or commented out) and the code should run without any errors. 

Hope my reasoning makes sense (and correct).  
I was struggling with this all afternoon and finally made some progress.

AchrafXVV, 2016-11-27:
Thank you so much @dtheva ! You saved my time !

candanerim, 2017-02-13:
Hi @dtheva.. 
I removed the initDevice function from the code but now I met the error: 
" There no Kinect devices, returning null. Could not open camera: -12
Failed to open camera subdevice or it is not disabled.Failed to open motor subdevice or it is not disabled.Failed to open audio subdevice or it is not disabled.
"
I use v1473 and Windows10. Can you help me to solve this problem?

AbdullahAslam, 2017-03-28:
@candanerim  i'm facing same issue.. plz if u solved this issue plz guide me.
Thanks

TomAuger, 2017-07-07:
If you can't init the device using initDevice() (v1), then initVideo() or initDepth() instead, depending on which one you want. If you want the IR video image, this becomes:

initVideo();
enableIR(true);

TheGuardiaan, 2017-11-12:
@candanerim try use Zadig to install the libusbK driver for each device you wish to use. This includes installing the libusbK for all the kinect v1 drivers listed on Zadig GUI.

https://github.com/shiffman/OpenKinect-for-Processing

adrianhoodmech, 2019-02-26:
@TheGuardiaan I'm pretty new to all this including Zadig, so following your suggestion took some experimentation. For the sake of exposition, here's what I did after having the same issue as @dtheva and following your instructions:
1. Open Zadig and click under Options > List all devices
2. In the main selection box, select XBox NUI Camera, then in the box where it says USB Serial CDC, click the down arrow until it pulls up libusbK (v3.0.7.0)
3. Click Replace Driver, and once completed this should resolve the issue.

adrianhoodmech, 2019-02-26:
@TheGuardiaan I'm pretty new to all this including using Zadig, so following your suggestion took some experimentation. For the sake of exposition, here's the steps I took after having the same issue as @candanerim and following your instructions:

1. Download/run Zadig and click under Options > List all devices
2. In the main selection box, select XBox NUI Camera, then in the box where it says USB Serial CDC, click the down arrow until it pulls up libusbK (v3.0.7.0)
3. Click Replace Driver

Once complete, your computer should now recognize the Kinect device and display the depth data. Thanks!

arturoherran, 2020-07-18:
This video by programafacil is very useful to use Zadig, and even though it's in spanish you will be able to understand what you need to do. It worked for me and started using my Kinect v1 right away. Hope it works.  

https://www.youtube.com/watch?v=o5K3boj8bVw&feature=emb_title

> @candanerim try use Zadig to install the libusbK driver for each device you wish to use. This includes installing the libusbK for all the kinect v1 drivers listed on Zadig GUI.
> 
> https://github.com/shiffman/OpenKinect-for-Processing