shiffman/OpenKinect-for-Processing #84: Get scaled down images to speed up processing

wassgha, 2016-08-18:
Hello,

Is there any way to specify the Kinect resolution for the RGB image on the Kinect v2? I am trying to process the image with OpenCV and it is taking too long (totally not adapted to live processing). I tried to resize the image before passing it to OpenCV (using img.resize) but I ended up with a weird striped image:

[image]

Thank you in advance!

ThomasLengeling, 2016-08-18:
Hello wassgha, do you have part of the code that I could look at?

wassgha, 2016-08-18:
To put you in context, I am trying to use OpenCV's calibration (using a checkerboard) on the video image from the Kinect v2 (I am trying to make a projector-kinect touch interface)

`import org.openkinect.freenect.*;
import org.openkinect.freenect2.*;
import org.openkinect.processing.*;
import gab.opencv.*;

class PWindow extends PApplet {
  Kinect2 kinect2;
  ArrayList<PVector> cornerPoints;
  OpenCV opencv;
  PImage video;
 
  PWindow() {
    super();
    PApplet.runSketch(new String[] {this.getClass().getSimpleName()}, this);
  }

  void settings() {
    size(1024, 848);
  }

  void setup() {
    background(255);
    kinect2 = new Kinect2(this);
    kinect2.initVideo();
    kinect2.initDepth();
    kinect2.initIR();
    kinect2.initRegistered();
    // Start all data
    kinect2.initDevice();
  }

  void draw() {
    background(255);
    video = kinect2.getVideoImage();
    video.resize(kinect2.colorWidth/5, kinect2.colorHeight/5);
    image(video, 0, 0);
      opencv = new OpenCV(this, video);
      cornerPoints = opencv.findChessboardCorners(9,6);
      for(PVector p : cornerPoints){
        ellipse(p.x, p.y, 5, 5);
      }  
  }
}`

bradenneufeld, 2016-08-18:
We experienced a similar issue during calibration - if you use kinect.getRegisteredImage() instead of kinect2.getVideoImage() performance is considerably improved and the calibration is accurate. The registered image has the depth sensor's resolution with the colors mapped to the depth "pixels".

wassgha, 2016-08-27:
I get the same problem even with `getRegisteredImage()` :(

abdelwahed, 2018-10-17:
I have the same issue