shiffman/OpenKinect-for-Processing #88: Mac OS X "No Kinect devices found"

manorch, 2016-09-07:
> "No Kinect devices found.
> Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.There are no kinects, returning null"

Downloaded freenect just in case too.

Running El Capitan, Processing 3.1.2, and using Kinect V1 with a power cord. 

The Kinect seems to act the same with and without the power cord plugged in and the green light seems to flicker at different intervals. Could it be a power cord problem?

The device also isn't showing up under USB in system diagnostic.

Any help is really appreciated.

enzyme69, 2016-09-19:
Check your Kinect number, maybe from the bottom? Have you tested using Kinect 1 or 2 PDE script?

manorch, 2016-09-20:
I tried using the Kinect 2 script and it didn’t work either.

It’s definitely the first model.

It’s strange, it started working randomly and would shut off after 2 mins. 

Sometimes when I open System Information it will not list the kinect, sometimes it will list all three kinect inputs, today it’s only listing the motor. It has also decided not to work today.

Totally at a loss of what to think.

carloscastellanos, 2016-10-13:
Same issue. Using Kinect V2, Running 10.10.5, Processing 3.1.2

Tried several of the examples including AveragePointTracking2 and RGBDepthTest2

The Kinect is however showing up under USB in system report.

sebmorales, 2016-11-05:
Any updates on this? 
I was trying with the kinect 1414 as well as the 1473 using the kinectv1 lib but kept getting the same error. 

"No Kinect devices found.
Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.There are no kinects, returning null"

The kinect did appear as connected when typing "/usr/sbin/system_profiler SPUSBDataType" in terminal.

manorch, 2016-11-05:
No updates on my front I kind of gave up trying to use the kinect with my computer

PitOgermann, 2016-11-27:
I had the same problem. And I have not solved it yet. But I have a part solution: 

It works with my MacBook Pro (Retina, 13-inch, Mid 2014) only when I do not connect it directly to it´s USB 3.0 Port but connect it to my Apple (27 Zoll) Cinema Display USB port with USB 2.0 and then connect it to the MacBook. 

So I think the Problem is the USB 3.0 to USB 2.0 connection. Is there a possibility to use a Adapter to force the MacBook using USB 2.0? 
Or is there a software solution to force the software using USB 2.0 instead of USB 3.0?  

**Thank for your Help!**

PitOgermann, 2016-11-27:
**Connections with connecting MacBook to Monitor and USB 2.0:**
[https://files.mycloud.com/home.php?brand=webfiles&seuuid=062b301d0d4dd26682702bdbc11d4501&name=ConnectedToUSB2](url)

**Connections connecting Kinect directly to USB 3.0 Port:**
[https://files.mycloud.com/home.php?brand=webfiles&seuuid=e95ee03a5179774f34bd0d27d04f8470&name=ConnectedToUSB3](url)

Michelleyaooooo, 2018-03-14:
I have the same issue. BTW the two links above are 404. So is there issue for Mac's USB 3.0 ? We should try to use USB2.0 instead?

PitOgermann, 2018-03-14:
I solved the problem for my project with a hardware solution. So I bought the cheapest usb2 hub that I could find. When connecting this to the Kinect instead of directly plug it into the Mac the Kinect is detected by the Mac.

saulomarc, 2018-03-22:
Hi, can you give the exact brand of the usb2 hub you used? @PitOgermann

NorielSotto26, 2018-03-22:
Hi, which kinect model are you using, The 1414 or the 1473? 

If its the 1473 model, I thought it can only support USB 3.0s? Or is it really effective using the 2.0 hub?

I just badly needed my kinect to work. I have 1 month left before my presentation. Thanks!