shiffman/OpenKinect-for-Processing #89: UnsatisfiedLinkError

scratchyone, 2016-09-09:
I get this error: ```
UnsatisfiedLinkError: org.openkinect.freenect2.Device.jniInit()J
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
```
I'm using ubuntu 16.04 LTS. I get this when running `kinect = new Kinect2(this);`

esreynaa, 2016-09-20:
If you haven't already, install the ARM Linux version of OpenKinect from [here](http://sukzessiv.net/%7Egohai/p5-arm/processing-openkinect-arm-fixes-4d25eb2.zip). Then open up an example for v1 Kinect version and don't forget the 51-kinect.rules change.

Davy-Huang, 2017-03-26:
If you were able to resolved this issue, can you explain in more detail how this was fixed?

closed 2020-12-11 by scratchyone