shiffman/OpenKinect-for-Processing #91: Kinect contour silhouette

pablovitamin, 2016-09-20:
Hello, Is it possible to have just the contour of the visible depth. I mean just the contour of some person standing in front of the kinect.

RORVI, 2016-11-20:
I'm working on something like this right now ! As far as I know, in the Kinect documentation there is mentioned something about the user pixels. And you can manipulate these ! So, for example, if you only need the silhouette contour, you should check if the pixel you're checking is surrounded from all 8 sides by other silhouette pixels. If it is, then it's not part of the contour, otherwise it is ! I'll let you know more once i solve this ! Good luck !

smokhov, 2016-11-20:
It is absolutely possible and is being used for example in our Illimitable Space System when we put some of the contours as parts of the visual effects on stage during live performances, e.g., here:

- https://vimeo.com/141811579
- https://vimeo.com/152476114
- https://vimeo.com/166999333

You'd have to use _OpenCV for Processing_ (just install it from the libraries menu of Processin) and feed it the depth data from Kinect and call the appropriate contour finding function of OpenCV; it has examples that just do that.

However, this is hardly a question for this library :-)

SteveEugene, 2017-08-20:
smokhov is this compatable with the current Mac OS? I would love some help with this if so!

smokhov, 2017-08-21:
On Sun, 20 Aug 2017, SteveEugene wrote:

> smokhov is this compatable with the current Mac OS? I would love some help with this if so!

Yes, our production laptop that ISS runs on for the actual shows is OS X, 
actually. We did a couple of those this year, e.g.,

   https://www.youtube.com/watch?v=yjH0S7Otwj4

so yes, it runs on the latest OS X, not sure I'd have time to help though 
as my ET Summit is upcoming, but you can follow through the OpenCV 
examples and just give it the depth image.

-- 
Serguei Mokhov, PhD, Manager, Networks & Security, AITS | /~\    The ASCII
Affiliate @ Computer Science and Software Engineering & | \ / Ribbon Campaign
Concordia Institute for Information Systems Engineering |  X    Against HTML
Concordia University, Montreal, Quebec, Canada          | / \      Email!
http://www.cs.concordia.ca/~mokhov
---
Co-founder, CCIFF, http://cciff.ca, 2nd ed., September 23-27, 2017
Call for submissions, ET Summit: http://cciff.ca/2017/festival/et/
---
Technical Lead
http://mdreams-stage.com
http://users.encs.concordia.ca/~gipsy
http://marf.sf.net | http://sf.net/projects/marf