shiffman/OpenKinect-for-Processing #92: i'm getting this error :(

maria4lexy, 2016-10-02:
64 windows
There are no kinects, returning null
Could not open camera: -12
Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.

OmegaNalphA, 2017-01-31:
If you are using Kinect1, there is no need to use kinect.initDevice(); try that it may help

AbdullahAslam, 2017-03-28:
@maria4lexy  : hey i'm having same issue... There are no kinects, returning null. 
have u sort out the problem > if yes plz guide me.
instead kinect.initDevice();  write  kinect.activateDevice(0);
it worked for me... but still getting error There are no kinects, returning null.