shiffman/OpenKinect-for-Processing #93: Syphon

juanmartin, 2016-10-04:
Hi there, I made a PImage using getRawDepth and only taking pixels within certain threshold as Daniel's video tutorial. I got this working fine.
I need to send the image through Syphon but it doesn't seem to refresh the image on the receiving side. My sketch is outputting the image fine, but I tried to pick the syphon from Resolume, VDMX, another syphon capturing example sketch, but none seem to refresh the image.

This is the kinect-syphon-sending sketch
[image]

This is what I get from syphon in another sketch (which picks up the first syphon server)
[image]
This last one changes according to... I don't know what, but it always displays some random glitchy stuff.

This is the console output from the Kinect sketch:

> 64 mac
> :/Applications/Processing.app/Contents/Java/core/library:/Users/juanmartin/Documents/Processing/libraries/Syphon/library:/Users/juanmartin/Documents/Processing/libraries/openkinect_processing/library:/Applications/Processing.app/Contents/MacOS/v2/mac
> Found path: /Users/juanmartin/Documents/Processing/libraries/openkinect_processing/library/v2/mac/
> Init Kinect2 
> [Freenect2Impl] enumerating devices...
> [Freenect2Impl] 22 usb devices connected
> [Freenect2Impl] found valid Kinect v2 @20:7 with serial 075529740447
> [Freenect2Impl] found 1 devices
> 1 Device Connected!
> [OpenCLDepthPacketProcessor::listDevice]  devices:
>   0: Intel(R) Core(TM) i7-3615QM CPU @ 2.30GHz (CPU)[Intel]
>   1: HD Graphics 4000 (GPU)[Intel]
>   2: GeForce GT 650M (GPU)[NVIDIA]
> [OpenCLDepthPacketProcessor::init]  selected device: HD Graphics 4000 (GPU)[Intel]
> Devce: 0
> [Freenect2Impl] enumerating devices...
> [Freenect2Impl] 22 usb devices connected
> [Freenect2Impl] found valid Kinect v2 @20:7 with serial 075529740447
> [Freenect2Impl] found 1 devices
> [Freenect2DeviceImpl] opening...
> [Freenect2DeviceImpl] opened
> Enable Kinect 2 Depth
> [Freenect2DeviceImpl] starting...
> [Freenect2DeviceImpl] ReadData0x14 response
> 92 bytes of raw data
> 0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
> 0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
> 0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
> 0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
> 0x0040:  31 33 00 00 00 0b 0d 0d 47 4b 53 33 34 39 2e 31    31332e2e2e2e2e2e474b533334392e31
> 0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e
> 
> [Freenect2DeviceImpl] ReadStatus0x090000 response
> 4 bytes of raw data
> 0x0000:  7f 26 00 00                                        7f262e2e
> 
> [Freenect2DeviceImpl] ReadStatus0x090000 response
> 4 bytes of raw data
> 0x0000:  7f 26 00 00                                        7f262e2e
> 
> [Freenect2DeviceImpl] enabling usb transfer submission...
> [Freenect2DeviceImpl] submitting usb transfers...
> [Freenect2DeviceImpl] started
> Device Serial: 075529740447
> Device Firmware: 1.1.3323.0.7
> Syphon 2.0-RC2 by Andres Colubri http://andrescolubri.net/
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 0
> 2016-10-03 22:01:11.771 java[8153:641564] SYPHON DEBUG: SyphonServerConnectionManager: Start Connection
> 2016-10-03 22:01:11.772 java[8153:641564] SYPHON DEBUG: SyphonServerConnectionManager: Created connection with UUID: info.v002.Syphon.ECF69F7A-6FA8-4557-BCBE-E5E9E7D49F85
> OpenGL error 1282 at top endDraw(): invalid operation
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 991
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 1022
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 959
> 2016-10-03 22:01:17.284 java[8153:641434] SYPHON DEBUG: SyphonServer: Got Discovery Request
> 2016-10-03 22:01:17.302 java[8153:641434] SYPHON DEBUG: SyphonServerConnectionManager: Add info client: info.v002.Syphon.40C111D1-50B4-4BB1-BB42-56E34C70BEAD
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 991
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 959
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 895
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 959
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 959
> [DepthPacketStreamParser::onDataReceived] not all subsequences received 991
> 2016-10-03 22:01:24.584 java[8153:641466] SYPHON DEBUG: SyphonServer: Retiring a server at code unload time because it was not properly stopped

These:
> [DepthPacketStreamParser::onDataReceived] not all subsequences received XXXX

Are pretty annoying too.

I'm posting this here because I tried some other Syphon examples and they work fine but only when I use this kinect library this happens.

Thank you in advance!
JM

juanmartin, 2016-10-04:
OK, seems I had to post something for my brains to figure it out! Fastest auto-response-solution ever.

**THIS** solved it:
  PJOGL.profile=1;
in my settings function.

After I realized about it I made a quick search and this came out: https://github.com/Syphon/Processing/issues/25

Well, hope this helps somebody!
Sorry if it wasn't this the correct place to post this, I'm still discovering stuff!
Cheers
JM

closed 2016-10-04 by juanmartin

yveacst, 2019-06-03:
Hey juanmartin,
Since OpenNI is no longer avaiable how did you get to run open kinect into syphon?
Im trying to use it in Resolume Arena 4

Thanks

juanmartin, 2019-06-04:
I use the PImage generated by the library and send it through the Syphon object. Remember to import both libraries.