shiffman/OpenKinect-for-Processing #96: Issue of Kinect for Windows not supprting openKinect library

Neeshad, 2016-10-14:
Hi ,
I'm  using kinect for one of our projects, where i have to make an interactive wall.
The major problem I'm facing now is openKinect library is not supporting Kinect for Windows.I referred so many  tutorials of you but could not proceed further with this major issue,and installed libusb as well,but still not working.I would be much grateful to you if u can give me a solution as soon as possible

chambefort, 2016-11-11:
Did you find a solution ?

sener07, 2017-10-08:
i have the model 1517 i am also having some problems : 
$ freenect-glview
Kinect camera test
Number of devices found: 1
Found sibling device [same parent]
Failed to set the LED of K4W or 1473 device: -1
Found sibling device [same parent]
Could not open device: -1
Could not open device

josuerocha, 2017-12-28:
Did you find a solution?

skribtsov, 2018-02-21:
I also need a solution for this....

skribtsov, 2018-02-21:
how to use fwfetcher.py ?