shiffman/OpenKinect-for-Processing #99: Build library from source

quinkennedy, 2016-11-17:
I would like to try adding support for the Kinect 2 on Linux, but I don't know how to build this library. Are there instructions somewhere (ideally for command-line building)?

noonker, 2020-10-01:
I'm also curious if there's instructions for this somewhere

klopi-wtal, 2020-10-09:
Also tried to make the Kinect v2 working with Linux (Manjaro). In the latest repository ist no linux-directory in sketchbook/libraries/openkinnect_processing/library/v2. Here how I build the Library from the master:

First I downloaded the master form this git, then edited /Openkinect-Processing/recourrces/build.properties;
line 33 from: _classpath.local.location=/Applications/Processing.app/Contents/Java/core/library_ to _classpath.local.location=/opt/processing/core/library_
AND the sketchbook location (line 20) from sketchbook.location=${user.home}/Documents/Processing to sketchbook.location=${user.home}/sketchbook

Then compiled with `ant -f build.xml`

The Library ist automatically copied to the /v2/linux libraries folder, the AddLibrary tool reconize it as Version 1.1 running the kinect v2 example leads to:

_java.lang.UnsatisfiedLinkError: Can't load library: /home/username/sketchbook/libraries/openkinect_processing/library/v2/linux/libfreenect2.so_

It does not work because of the missing file _libfreenect2.so_ in /openkinect_processing/library/v2/linux. I copied the missing file from /usr/lib and got this error:

_java.lang.UnsatisfiedLinkError: org.openkinect.freenect2.Device.jniInit()J
UnsatisfiedLinkError: org.openkinect.freenect2.Device.jniInit()J
A library relies on native code that's not available.
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:408)
UnsatisfiedLinkError: org.openkinect.freenect2.Device.jniInit()J
	at java.lang.Thread.run(Thread.java:748)
Or only works properly when the sketch is run as a 32-bit application._

Anybody an idea whats wrong?