shiffman/OpenKinect-for-Processing #104: failed to open KinectV2

OmegaNalphA, 2017-01-31:
I continually get this error when trying to run the initial program 

64 windows
Init Kinect2 
[Freenect2Impl] enumerating devices...
No Device Connected!
[Freenect2Impl] 8 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @1:18!
[Freenect2Impl] failed to open Kinect v2 @1:18!
[Freenect2Impl] found 0 devices
Cannot Find Devices

I have seen the solutions for linux/mac, but does not seem to be a solution for a windows system. Can anyone help?

paulhiggs, 2017-02-09:
I get the same under Windows with a Kinect v2. The Processing console has ...

64 windows
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @1:9!
[Freenect2Impl] found 0 devices
No Device Connected!
Cannot Find Devices

Other Kinect apps run OK (Kinect Studio, Brekel) run OK and lsusb give the following
```
C:\Users\paul_>lsusb -help
lsusb

VendorID:0461 ProductID:4d17  "USB Input Device"   "USB Optical Mouse"
VendorID:064e ProductID:c355  "USB Composite Device"
VendorID:8087 ProductID:0a2a  "Intel(R) Wireless Bluetooth(R)"
VendorID:04f3 ProductID:2274  "USB Input Device"
VendorID:045e ProductID:02c4  "USB Composite Device"   "Microsoft"    "Xbox NUI Sensor"    "004925160847"
```

Yaffa1607, 2017-05-01:
Did you find a solution for  this ?

paulhiggs, 2017-05-01:
@Yaffa1607 - I have no solution (except to stop using Processing under Windows)

paraclete-pizza, 2017-08-08:
I am having the same issue. Windows 10, Kinect 2. The Kinect works fine with other programs, and even with Thomas Lengeling's Kinect V2 Processing Library (which unfortunately cannot align the color and depth images, or I'd just use that). But this library claims it cannot detect the Kinect.

OmegaNalphA, 2017-08-08:
Yeah I do remember Zadig working well for me, and it had solved a lot of my issues. This is a pretty good library for making some quick games but stops short with the depth matrix. Honestly what ended up solving my issue after using Zadig was to just unplug it and plug it back in. I suspect its an issue with detecting new devices using this library on windows. I have seen this pop up from time to time on Macs as well however.

OmegaNalphA, 2017-08-08:
This may help as well: https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting

paraclete-pizza, 2017-08-08:
I also found that the examples in the library Processing 3 downloads automatically are out of date, and replacing them directly with the ones from GitHub fixed my issues with the examples running.

So Zadig does get this library up and running, and well. But having to change your USB drivers effectively kills the Kinect for other uses (until you swap drivers back), and keeps you from using other Kinect libraries (which is an issue when this library doesn't do skeleton tracking)...

Still, fun to play with. But I may need to brush up on my C++ and try messing around in OpenFrameworks to do more sophisticated stuff...