shiffman/OpenKinect-for-Processing #106: What does 'Isochronous transfer error: 1' mean?

louiserobson, 2017-02-01:
I'm running openkinect with a Kinect v1 1414 on Windows 10. For a start my Kinect only seems to work occasionally with the examples if it is plugged into my USB 3.0 port, if it's plugged into either of the other USB 2.0 ports it doesn't work at all. 

When I try and run the openkinect examples, they work maybe 50% of the time at varying levels - mostly, whatever image is called loads fine, but the keyPressed functions do not work. 

Every time I run any of the example sketches I continuously get 'Isochronous transfer error: 1' in the console, I think it's being returned every single frame - what does this mean, and how can I fix it? I think it must be something to do with memory?

candanerim, 2017-02-13:
Hi, I met the same error "Isochronous transfer error: 1" when I tried to run Shiffman's [https://github.com/shiffman/OpenKinect-for-Processing/tree/master/OpenKinect-Processing/examples/Kinect_v1/AveragePointTracking] example code for Kinect. I have Kinect v1 1473 and Windows 10 and downloaded drivers with Zadig. I searched the forums, there are so many people who met this error but still couldn't fix. If you find a solution please let me know.

AbdullahAslam, 2017-03-28:
@candanerim @louiserobson  found the solution yet ? 
if yes please help me.

candanerim, 2017-03-29:
I couldn't find any solution. @AbdullahAslam

jnydutra, 2017-09-19:
Downgrade libusbK worked for me. Use http://zadig.akeo.ie/downloads/zadig_2.0.1.exe on windows.