shiffman/OpenKinect-for-Processing #109: Processing .exe does not work with Kinect

ClaraBellinMMI, 2017-03-08:
I have a program that works with a Kinect (v1 model 1414), I did a tracking point. When I export the program, and then launch it as an .exe it does not work. A window comes but it's only grey and then my pc freeze.

I've seen that it's possible to change things in some .jar files, but I don't know which one

nevahid, 2017-03-09:
hi there Clara,
i've met with this problem before and found out a solution. find below files on the net (and even in this github) and paste them in OpenKinect_Processing folder, inside library folder (there is a an "openkinect_processing.jar" file over there)
1- jna.jar
2- jna-platform.jar
3- jnidispatch.dll
wish you success

ClaraBellinMMI, 2017-04-05:
Sorry for the wait, I totally forgot to answer.
But this works perfectly, thank you !

closed 2017-04-05 by ClaraBellinMMI

DannyRoberts95, 2018-07-02:
Hey there guys! I am having a similar problem but can't seem to fix it. 
What exactly did you guys do to sort out the problem? Where did you find the files? Where in the folder hierarchy did you place them?

Thanks folks!