shiffman/OpenKinect-for-Processing #110: Not compatible with the current OS or is a 32 bit system

adilbonzi, 2017-03-19:
I recently installed processing 3.3 on my Ubuntu 14.04 LTS. I also was able to successfully interface with arduino UNO. Now when Im trying to run my kinect v2 in processing 3.3, it is giving me the following error:
 
   not compatible with the current OS or is a 32 bit system
    UnsatisfiedLinkError: org.openkinect.freenect2.Device.jniInit()J
    A library relies on native code that's not available.
    Or only works properly when the sketch is run as a 32-bit application.

This is the code that i entered for so far:
import org.openkinect.processing.*;

Kinect2 kinect2;

void setup(){
  
  size(512,424);
  kinect2 = new Kinect2(this);
  
  kinect2.initDevice(); 
}

void draw(){
  background(0);
}

Any suggestions on what procedure is to be done.

bergbuehne, 2017-04-01:
Same problem with
Linuxmint 18, Processing 3.3
in Line: kinect2 = new Kinect2(this);
What can i do?

bergbuehne, 2017-04-01:
Hallo,
sorry for my english, it's not my native language.
It works now.

First i do this: [https://www.reddit.com/r/openkinect/comments/5e9muc/installing_on_linux_mint_18_sarah_mate/](url) 

Then i changed all "Kinect2" to "Kinect".
And change the Line:  kinect2.initDevice(); with: kinect.initVideo();
It's from the Example-files.
Now it run.
If i change it again, "Kinect" with "Kinect2", generates the Error again.

Regards

jav-solo, 2019-02-20:
I am having this same issue